fmalpartida/New-LiquidCrystal #3: Getting Compile warnings about "'boolean' is deprecated" when compile Bigtreetech SKR PRO V1.1 board for Marlin-bugfix-2.0

GadgetAngel, 2020-01-30:
I am trying to compile Marlin-bugfix-2.0 for a Bigtreetech SKR PRO V1.1 that uses the New LiquidCrystal Library and I am able to get a successfully compile but I get WARNINGS from the LiquidCrystal routines and I thought you should be aware of these warnings:
[image]
[image]
[image]
[image]
[image]

elektronikworkshop, 2020-02-11:
Hi!

I stumbled over your issue by accident: Francisco will review my [pull request](https://bitbucket.org/fmalpartida/new-liquidcrystal/pull-requests/12/fix-issue-93-and-migrate-boolean-to-bool/diff) which fixes those issues (and other warnings) next weekend. I'm not sure if he syncs the GitHub repository. Better you watch the pull request and get a new release from there.

Regards/EW

Djstephenson1, 2020-09-30:
Was this ever resolved? I am experiencing this

GadgetAngel, 2020-09-30:
> Was this ever resolved? I am experiencing this
 Yes, I still get warnings:
[image]
[image]
[image]
[image]
[image]
[image]

They are just warnings. I only made an issue of it to make the programmer aware of the issue.

fmalpartida, 2021-06-02:
Hi folks, for some reason the pull request is not accessible or no longer there.
If not, I will make the updates and merge into the integration branch at some point for folks to try out.