fmalpartida/New-LiquidCrystal #6: Library not available in Arduino IDE library manager

jackjansen, 2021-01-03:
I've been using this library (in stead of the original one) for a long time, and very happy with it.

But now that I'm moving many of my projects over to GitHub Actions as Ci/CD system it's bothering me more and more that the library cannot be installed automatically through the Arduino IDE or `arduino-cli`.

The library does feature in the patlformio library registry, so for platformio builds things do happen automatically, but not for Arduino.

Could you register your library with the Arduino library manager, please?

FastThenLeft, 2021-10-04:
I second that! I use the library in a class and it would be great if the students could use the standard library manager in the standard Arduino IDE.

jackjansen, 2021-10-04:
@fmalpartida if you haven't added a library to the Arduino library registry before: it's pretty simple: just fork <https://github.com/arduino/library-registry>, add your library, submit a pull request. (after reading the instructions over there on the structure your repo should have, but I think it already has it).

Or I could do it (but only with your permission, otherwise it would sort-of feel like I'. stealing the credits for your work)

per1234, 2024-01-15:
Here is what would be required to make the library available for installation via the Arduino Library Manager:

1. Add a `library.properties` metadata file:
   https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format
1. Create a new tag or release in the library repository.
1. Submit the library to the Arduino Library Manager registry (as already mentioned by **jackjansen**):
   https://github.com/arduino/library-registry#adding-a-library-to-library-manager

If you are interested in doing this and have any questions or problems, please feel free to let me know and I'll provide assistance.