fmalpartida/New-LiquidCrystal #7: can I use 3 displays as an array

muuc, 2021-02-24:
I am using your library, thanks for creating.

until now:
#include <Wire.h> 
#include <LCD.h> 
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd5(0x25, 2,1,0,4,5,6,7,3,POSITIVE);
LiquidCrystal_I2C lcd6(0x26, 2,1,0,4,5,6,7,3,POSITIVE);
LiquidCrystal_I2C lcd7(0x27, 2,1,0,4,5,6,7,3,POSITIVE);
...
lcd5.begin(16,2);
lcd6.begin(16,2);
lcd7.begin(16,2);
...

Is it possible to create an array of lcds to use sth like
for (byte i=0; i<3; i++) lcd[i].begin(16,2);

Thanks in advance!

fmalpartida, 2021-02-24:
Hi,

Thanks for using the library. 

Yes, it should be possible. They would be independent instances that you can the us. 

Best,
fm

iPhone courtesy typos

muuc, 2021-02-24:
Hi,

thanks for your quick response.

after many trials and errors ...

I succeeded with

LiquidCrystal_I2C lcd_5(0x25, 2,1,0,4,5,6,7,3,POSITIVE);
LiquidCrystal_I2C lcd_6(0x26, 2,1,0,4,5,6,7,3,POSITIVE);
LiquidCrystal_I2C lcd_7(0x27, 2,1,0,4,5,6,7,3,POSITIVE);

LiquidCrystal_I2C lcd[3]= {lcd_5, lcd_6, lcd_7};
...
for (byte i=0; i<3; i++)  lcd[i].begin(16,2);
...

Is this the only way?

kind regards
H.Demmer

fmalpartida, 2021-06-02:
Very much it should be. The way the library was written was to enable users to create multiple instances of the LCD.
However, things are not written to multiple LCDs as if it was only one LCD.
You have to manage what you present on each individual LCD. In the example above, address lcd_5, lcd_6, lcd_7 with the corresponding methods.

closed 2021-06-02 by fmalpartida