fmalpartida/New-LiquidCrystal #9: analogWrite

owenduffy, 2021-04-15:
The library REQUIRES analogWrite, even if not called.

ESP32 Arduino does not have an analogWrite function and the lib will not compile.

It would be helpful if there was come compile time conditional code to implement either analogWrite code (as exists) or to treat the backlight pin as a digital pin (with or without inversion etc).

Owen

Dlloydev, 2021-04-15:
I've just created a small analogWrite library that I've tested OK using an ESP32 dev board .... (awaiting for delivery of an ESP32-S2 board) so an update might be necessary. Works with 8 PWM and 2 DAC pins ... resources are automatically managed.

[image]      [documentation](https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite)

owenduffy, 2021-04-15:
Re the lib, I did try that for my migration of an ESP8266 project to ESP32 to satisfy the "analogWrite not found" problem, but it broke the build with a lot of error messages. At that point, knowing it was not a functional requirement, I aborted the migration.

The fundamental issue with NewLiquidCrystal (which is an excellent lib, great work), is that analogWrite is only needed if you want to control backlight brightness proportionally, and since I am using the i2c variant, that is not possible anyway (AFAIK).

So, it would be useful to be able to conditionally replace the analogWrite  statements with simple digital pin control at the user's option.

I could simply provide an empty analogWrite function in my code to satisfy the build... but the solution suggested above would be better... I think.

Owen

Dlloydev, 2021-04-15:
Huh? Are you sure you've already tried the correct one? 
Note that my library is called **ESP32 ESP32S2 analogWrite** (click the badge or link above)

There's already someone else's library called ESP32 AnalogWrite (this one is not mine).

Note that I'm awaiting delivery of a ESP32-S2 board for testing, but this is code is fully tested, compiles and works OK on a ESP32 Dev Board.

owenduffy, 2021-04-15:
I tried the other one, ESP32 analogWrite. Sorry for the confusion.

Dlloydev, 2021-04-15:
That's fine .... unfortunately, the most ideal name for the library had already been taken.

owenduffy, 2021-04-17:
I thought I had update the lib, and I had not... apologies, the latest lib works fine.

closed 2021-04-17 by owenduffy