fmalpartida/New-LiquidCrystal #13: Optimizations for I2C library?

RobTillaart, 2023-12-21:
Hi,

First, thank you for your excellent library!

As I was in the need for speed, I investigated if I could write a faster I2C liquid display library.
After a week the code (0.1.2) is reasonable stable and performance is meeting my needs.

It might be interesting to look at some of the optimizations I made to include in your I2C part.
Most important the library send a byte in one transaction of 5 bytes instead of using 4 transaction (one per nibble).
The PCF8574 handles this nicely.
Furthermore as the 4 data pins are in ascending order one can prepare the byte a bit faster.

I tried to keep the API similar to your library however I have not implemented all functions fully yet.
Finally I added a few new functions, e.g. **clearEOL()** and some more. 

The link is: https://github.com/RobTillaart/I2C_LCD and should be available through the library manager.

If there are questions, please let me know.

Regards,
Rob

closed 2024-12-30 by RobTillaart

fmalpartida, 2024-12-30:
Hi Rob, it’s been a rocky year and haven’t been able to pay attention to a lot of things. 

Thank you for using the library and sharing your fork. Very happy to read it has been useful. 

If you wish, you can process a pull request with your I2C changes into the main library for broader sharing. 

Happy New Year.

RobTillaart, 2024-12-31:
Hi,

The repo is not a fork, it is a rewrite from scratch while keeping the API (mostly) compatible, so a quick PR is not an option.
Rather busy so I don't know if I can find time to fork and PR on short term, but I'll make a note in my backlog.

Happy new year to you too!